threeme3/usdx #1: QCX resets after transmit and VOX not working

phckopper, 2020-01-03:
Hello, 
I've flashed my QCX with the 1.01s firmware and performed the hardware modifications detailed on the master branch (not the experimental one).

However, I'm having trouble using VOX and transmitting. When enabling VOX, it shows up as enabled, however the radio doesn't respond to sound being injected on the microphone. This might be some issue with my cabling, which I have yet to check, but I wanted to report anyway, in case it relates to the next problem:

Immediately after transmitting, my QCX resets even if VOX is not enabled. At first I suspected a power issue, but measuring the voltage and keeping the PTT pressed showed power was actually okay. Also, I can transmit indefinitely, the radio only resets when I release the PTT.

How do you thing I can troubleshoot my problems? How can I help solving these issues?

Thanks for the great work!
PY3PK

threeme3, 2020-01-03:
Hi Pedro,
Many thanks for trying out the experimental firmware and giving feedback on the issues you see. Much appreciated.

Can you tell me how many seconds the reset happens after releasing the PTT?

I think what happens is that the watchdog will reset the QCX due to some slowness while transiting from TX to RX.

73, Guido
PE1NNZ

threeme3, 2020-01-03:
Pedro,
I have uploaded a new firmware R1.01t, and it should resolve both issues you were seeing in R1.01s. 
I could reproduce your issue of always in TX when enabling VOX. Note that enabling VOX should be done with a long on right button (not via menu), the VOX sensitivity can be done in menu with VOX level. Let me know if this has also fixed the reset issue.

phckopper, 2020-01-03:
Hello,

Thanks for the quick answer! The update fixed the always in TX bug, however my QCX still resets after transmitting. I have recorded a video and uploaded it here: https://www.youtube.com/watch?v=0971YVNCNeo so you can see the issue. It doesn't seem to matter how long the PTT is pressed. Also, the QCX seems to switch back to receive before resetting (the R reappers on the bottom right shortly before reset). 

I also tried downgrading to 1.01d, however I get a bootloop on the initial splash screen, without any tests being run. Therefore, I believe this might be some watchdog related issue too. However, this seems weird as we have the exact same QCX (I guess).

Mine originally came with QRP Labs firmware T1.00f. My fuses currently are `(E:FC, H:D1, L:F7)` as read by avrdude. 

I try to mess around with your firmware as soon as I get it compiling here. I'll keep you updated.

Thanks again,
Pedro

phckopper, 2020-01-03:
**Update:** when compiling with debug enabled, I get !!CPU_tx=100.05% error on startup.

phckopper, 2020-01-03:
**Solved:** func_ptr still pointed to sdr_rx even when dsp mod wasn't detected on line 1055. Forcing it to dummy solved the issue, I guess. I still can't test it correctly as I'm away from home and can't check my biasing.

threeme3, 2020-01-04:
Hi Pedro, very good regarding your findings. Just to understand why you see this !!CPU_tx are you using Arduino 1.8.10? You are right regarding the wiring, it is probably this one that has an issue: `DVM/pin3-R57 to IC2/pin21 (AREF)`.

I am using fuse settings `(E:FD, H:D1, L:F7)`. Setting E:FC on my QCX will not let it start, probably because the supply voltage is below the Brown-Out detection Level (BOD) (see [10.5]), this also might also be causing your reset issue and maybe also why 1.01d gets into a boot loop.  

[10.5]: http://ww1.microchip.com/downloads/en/DeviceDoc/Atmel-7810-Automotive-Microcontrollers-ATmega328P_Datasheet.pdf

threeme3, 2020-01-04:
In the source code 1.01u I have added some debugging that can report if a brown-out reset has occured.

threeme3, 2020-01-05:
Another update: I have now updated from 1.8.9 to Arduino 1.8.10 too, and now also seeing  the !!CPU_tx=100.05% message. It seems that the new compiler generates less efficient code..

closed 2020-01-28 by threeme3