threeme3/usdx #2: Dead audio when using VOX with SSB mods

phckopper, 2020-04-13:
Hello,
I've finally made this project work on my QCX and performed only the SSB modifications. What I noticed is that, when VOX is enabled, the RX audio disappears (just noise), even when VOX isn't triggered (V showing on LCD, not transmitting). I'm running v1.02a and my RX audio works normally when VOX is disabled.

threeme3, 2020-04-14:
Hi Pedro,
Very good to hear you got the SSB mod working on your QCX! And many thanks for reporting on the VOX issue when only SSB mod is made. I have found the bug and re-uploaded the fix in R1.02a release, and it will solve your issue. 73, Guido

closed 2020-04-14 by threeme3

phckopper, 2020-04-14:
Thank you, Guido! Your mod is awesome and made my QCX spring back to life. You're awesome, if I could I'd buy you a beer :smile: