threeme3/usdx #3: Lcd flickers messed up, displaying random letters

ffanez, 2020-04-30:
Hello Mr. Guido PE1NNZ
Thank you very much for sharing this great program,    I want to try to follow what you have done but after VERSION "1.02a" upload was successful,   I got a messed up letter on the LCD, it doesn't work as it should,  can you give advice and help to solve it?  correctly.
Forgive my English is not good, thank you.

threeme3, 2020-04-30:
Hi ffanez,
Do you use the LCD display that is supplied by QRPLabs? or another display?

I can fix this if you tell me the exact LCD display type you are using.

73 Guido

ffanez, 2020-05-01:
I use LCD 1602 China shipments that I have in stock, but it only works well on the 1y version, I really appreciate your very fast response to my request, thank you very much, Mr. Guidos

threeme3, 2020-05-01:
Thanks. I have uploaded new firmware, can you please try? [](https://github.com/threeme3/QCX-SSB/releases/download/R1.02a/R1.02a.hex)

If not working, please let me know, there is more to solve.

73 Guido

ffanez, 2020-05-02:
Thanks,  Mr. Guido

I have tried uploading your new firmware, but it still seems unsuitable for my LCD display.

threeme3, 2020-05-02:
I have uploaded new firmware, could you check again?

If not working, I can try something else. Please let me know.
73, Guido

ffanez, 2020-05-03:
Hello Mr. Guido
I've tried it, always getting the same thing and again it hasn't worked.
Please, can you provide the recommended specifications / serialnumber parts and LCD brands for the firmware that you created,
or a suitable display replacement for me to be able to use well.
Some types have almost the same shape but don't work either.
I do not know the right type to use, because for the initial firmware version that you created can be displayed properly without any constraints, using the same LCD that I use.
Thank you for all your responses.

threeme3, 2020-05-06:
Hi ffanez,
I see you closed this issue, but I still trying to find a resolution for your display.
Did you fixed the issue with another display?
73, Guido

ffanez, 2020-05-06:
Hi Guido, my best regards
I am still waiting for some purchases of LCD type packages that I will try,
a little constrained by delivery because the corona virus pandemic might be
a bit long.
Of course I still really hope to be able to have QCX-SSB as good as you
have and always waiting for updates on its development.
I really want to try this great program to be realized, it's my dream right
now.
I'm sorry if I mistakenly closed the issue, well later I'll reopen, because
I'm afraid it will disrupt the core development of your program.
I am very happy for the response you have given, and fortunate to have
analyzed you here.
Thank you very much.

threeme3, 2020-05-08:
Hi ffanez,

I have changed way the LCD is driven. Can you test once again?
The firmware and code is here: https://github.com/threeme3/QCX-SSB/releases/tag/R1.02b

Many thanks, please let me know if is working or not working.
73, Guido

ffanez, 2020-05-08:
Hi Mr. Guido,
I've tried it and it works fine.  I am very glad it worked perfectly.
Thank you very much, your work is amazing.
73, ffanez

threeme3, 2020-05-08:
Hi ffanez!

Very good and many thanks for testing and reporting me back. I m very happy to hear it works now well.

73s and enjoy the trx

closed 2020-05-08 by threeme3