threeme3/usdx #5: Is the assembly PDF v 1.01 missing XTAL2 in checklist on page 2?

caseyjamesdavis, 2020-05-19:
Fantastic project!

Just starting my QCX-SSB build.

Cheers,
Casey
KC1JHV

threeme3, 2020-05-20:
Hi Casey!

Good catch. Manuel has updated the document and I have just uploaded it. Tnx!
Let us know if you have more findings.
Enjoy the build.
73 Guido

closed 2020-05-20 by threeme3

caseyjamesdavis, 2020-05-22:
Hi Guido!

I finished mine yesterday and it works great!

[image]

I do have a few more findings for the assembly manual.  What is the best way for me to share them with you (this gh issue, new gh issue, email, groups.io)?

73
Casey

threeme3, 2020-05-22:
Wow.. Great! Looks very good, and very happy it is working great.

Yes, the best is to pass the comments to Manuel DL2MAN, so he can take these findings in the next version of the assembly manual. You can send him privately in the forum. You can also pass the comments here and I will forward to him. Up to you whats the best for you.
Thanks gain, and enjoy your new transceiver :-)

caseyjamesdavis, 2020-05-22:
I will send directly to Manuel.

Thank you for sharing such an interesting and useful project.

73
Casey