threeme3/usdx #6: QCX+

14RubberDucks, 2020-05-29:
Will you be making a guide/new version for the new QCX+ transceiver. It seems like QRP labs are fully stopping the production of them and switching to the QCX+.

threeme3, 2020-05-29:
Hi. When the new QCX+ transceiver comes out, people who bought the kit might volunteer in writing an assembly manual. You are correct that right now there are no QCX available. 

Note that a new group with collaborative interest has started to build a compact, portable (SOTA) rig for the summer, based on the ideas presented here. More information can be found in this: http://ucx.groups.io

closed 2020-05-29 by threeme3

MauiKano, 2021-02-05:
i am in the process of modifying a QCX+. It's clear to me what to do and I have all mods implemented and tripple checked. I struggle however to test if it works. I can not receive anything. Just noise. Swiching modulation or antenna brings me different noise and different reception level, which is a good sign. But still, no QSO. Would love to talk to someone who has a working QCX+. I'll be happy to share my results.