threeme3/usdx #7: Freezes in TX with v 1.02e?

caseyjamesdavis, 2020-06-09:
Hi Guido,

I recently updated from the 5/17 v 1.02d QCX-SSB.ino to the 6/3 v 1.02e.  

After reboot, everything seemed good until I clicked the right button and then radio went in to TX and froze.  After another reboot it was still frozen in TX, and the frequency was missing from the display.  

The hardware is the Rev 5 QCX board built per Manuel's v 1.02 instructions.  I have not done the 'magic witchery mod'.  I also have a 10m LPF installed on the board.

I tried a few older commits between 6/3 and 5/17, but it did the same thing.

I am now back to the 5/17 commit and everything is back to normal.

Any ideas?

Cheers,
Casey

PS - Below are my first ever reports on 10m and FT8 thanks to your project!

[image]

threeme3, 2020-06-09:
Hi Casey,
Thanks for reporting this. Could you please check this again and see if this issue only happens in CW mode? If so, can you explain me what is connected to the keyer/mic jack?
Can you try if the issue is gone while disconnecting the keyer/mic input? 
Did you set pa bias Max to 128?
Thanks and 73,
Guido

threeme3, 2020-06-09:
Hi Casey,
I have made a fix in the latest code, can you check if the problem is resolved and re-open this issue when you still have an issue? Thanks again for reporting the issue and enjoy your new transceiver!
73, Guido

closed 2020-06-09 by threeme3

caseyjamesdavis, 2020-06-09:
Hi Guido,

Thanks for responding so quickly.

I have the 6/9 .ino commit installed and running.  No TX lock-ups observed.

It does display T when in CW mode as I cycle through the modes with a right button click.  The T remains until I go to a different mode.  I don't have anything connected to the keyer/mic jack.

Also, if I cycle through the modes via the menu, the encoder stops working when CW appears.

I have been keeping my PA Bias max under 100 to keep the transistors from getting hot during FT8 TX.  Is this a bad idea?

Thanks for your help,
Casey

threeme3, 2020-06-10:
Hi Casey,

Thanks for the feedback. 

Somehow on your unit the DAH keyer-line is continously keyed down (shorted to GND). 
This originally caused the lockup while navigating through the menu, and is causing the transceiver to go in TX when entering CW mode.

Can you check if you installed the pull-up resistor of 10K to AVCC in the DAH line (placed at D4 in the QCX) ?  You should measure 5V at the DAH line. Resolving this, will resolve the lock issue.

73, Guido