threeme3/usdx #10: 2m band

ok1cdj, 2020-08-18:
Hi, in source code is in band variable 2m frequency. Will the transmitter works in 2m band ? and how about receiving ? My idea is down converter to 28 MHz.

73 Ondra OK1CDJ

threeme3, 2020-08-18:
Hi Ondra!

Yes, the SI5351 is still working well on 2m and can produce a clock and modulation.
The BS170 and FST3253 stops working well above 6m and needs to be replaced by similar parts that can support <3.3ns switching times. Another idea is to use here a non-switch based type of mixer (e.g. a diode or dual-gate mosfet mixer).
It might be needed to put a pre-amp stage in-front of the FST3253 mixer to get more sensitivity.

Are you interested in a VHF only transmitter? 

73, Guido

closed 2020-08-18 by threeme3

ok1cdj, 2020-08-18:
Hi Guido, thank you for answer. I would like to make a simple TRX for 2m band based on your FW. Tx side is no problem as you said. I thinking about receiver now.

73 Ondra OK1CDJ

threeme3, 2020-08-18:
Hi Ondra, Here is the fast switch that might work well at 2m and can use as a drop-in replacement:  http://www.potatosemi.com/potatosemiweb/datasheet/PO3B3253A.pdf

ok1cdj, 2020-08-26:
Thanks Guido, just ordered some for experiment..

KajWiik, 2020-08-26:
Hi Guido and Ondrej,
I tried to order them also but it seems that Potatosemi does not sell
outside of the US? Any hints how to get them?

Thanks,
Kaj

sarma-mvs, 2020-12-13:
Hi Guido,
could you offer the firmware for only Tx at 2m . We can manage Rx with a n RTL SDR 
 I am much facinated and would like as SSB output for input and upconversion to 2.4GHz to use on Es'hail sat2  QO100.
 regards and seasons greetings

sarma
 vu3zmv