threeme3/usdx #22: VOX not enabled using menu

gmgaston, 2020-09-29:
Using 1.02h code

For a long time I was not able to get the VOX to work.  After troubleshooting and finding no problem,  I finally tried enabled VOX using the Right Button Long and then it worked.   So setting the VOX on using the menu does not seem to enable VOX, but the Right Button Long does.   I have low RF output but I know that is caused by the LPF and I can fix that.   Except for this issue,  everything is working great, working FT8 on 30 M.  Started with a new QCX+ and followed the modification instructions.   Thanks for all your work 

73, Greg
KD4VV

threeme3, 2020-10-07:
Hi Greg,
Yes, VOX function is currently only available with a long-right press. I will change this in near-future by moving the VOX function entirely in software. I will announce this in issue #23 as soon I have made the change.
Thanks for raising the issue, and enjoy your new rig :-)
Guido

closed 2020-10-07 by threeme3

threeme3, 2020-10-13:
Hi Greg,

There is a new version, where the VOX control is now entirely working from the menu (3.1). 
There is no need to use the long-right button press anymore :-).
When switched ON, you can freely tune around and control the rig.
The new version is here: [](https://github.com/threeme3/QCX-SSB/tree/feature-rx-improved)

Very interesting to read you have been modifying a QCX+. Would it be possible to share the modification instructions that you followed, so that others with a QCX+ can build?

Thanks again for your interest.
73, Guido

gmgaston, 2020-10-31:
Guido:
I did build my QCX-SSB  from a new QCX+ kit but I just followed the existing instructions  [PCB Rev5 Assembly Manual CLICK HERE] by Manuel, DL2MAN (document revision 1.1), 23-May-2020

The QCX+ has essentially the same schematic and all the part IDs  R1, D2, etc are the same as the original - seems only the layout is different.    I just followed the above instructions and referred the QCX+  parts layout to find the location and made all the substitutions outlined in the above file.  

I did use the QCX+ instructions for building the LCD board.   For the jumpers that Manuel  shows adding at the end - with the new layout those can't be done exactly as shown.   I wound up putting all the jumpers and the 2 resistors on the bottom side of the board so the top looks cleaner.  I used 30AWG wire wrap wire and some heat shrink.   I did also hot glue in 2 -   4 pin female headers  in the area where the LPF is supposed to built,  and used QRP-LAB's LPF kit which is on it's own little board so that I can swap out the LPF when changing bands.  

Hope this helps.   Thanks for all you do.   I am getting ready to install the most recent firmware.   

73
Greg
KD4VV