threeme3/usdx #23: Signal Meter Freezes when Vox = On

PE1DDA, 2020-10-02:
Whit VOX on, but no transmit "trigger" f.i. when using FT8, the signal meter freezes in all modes, both h and i version of software

threeme3, 2020-10-07:
Hi Cees,
Thanks for raising this issue. Yes, you are right the s-meter and other menu-functions are blocked as soon you enter VOX-mode. I am aware of this, and it is currently coded like that, however I am planning to move the VOX feature completely to the menu (so long right button press to switch-on VOX will no longer be there in near-feature), and VOX functions will be active simultaneously as other functions like s-meter and menu-functions, etc.
I will keep you posted here when it has been changed.
Guido

PE1DDA, 2020-10-08:
Hi Guido,

Thnx! I wil follow your developments on the foot as Louis van Gaal would
say :-)
Do you have a "beta version - so far" available for testing?

Cees

threeme3, 2020-10-08:
Yes! a beta version is here, but VOX is currently broken:

https://github.com/threeme3/QCX-SSB/blob/feature-rx-improved/QCX-SSB.ino

Note that on the top of the code, you need to change (enable) the following config settings for Barb's transceiver:

//#define QCX           1   // Disable for uSDX; enable to support QCX specific features (QCX, QCX-SSB, QCX-DSP with alignment-feature)  (disable this to safe memory)
#define DEBUG           1   // Hardware diagnostics (disable this to save memory)
#define KEYER           1   // CW keyer
//#define CAT           1   // CAT-interface
//#define F_XTAL 27005000     // 27MHz SI5351 crystal
#define F_XTAL 25004000   // 25MHz SI5351 crystal  (enable for uSDX-Barb or using a si5351 break-out board)
#define SWAP_ROTARY   1   // Swap rotary direction (enable for uSDX-Barb)
//#define OLED          1   // OLED display, connect SDA (PD2), SCL (PD3)
//#define TESTBENCH     1

PE1DDA, 2020-10-10:
Found something that may be of interest for you:
Firmware 1.02i with your suggested mods applied (below), receiver on 50 Ohm
terminator

*But maybe there is something in the PLL that is causing this. Could you
try to add the following line of code (at around line 1017, just above the
line "fvcoa = d * fout" ?      if(d % 2) d++; // add this line just above
the line belowuint32_t fvcoa = d * fout; // Variable PLLA VCO frequency at
integer multiple of fout at around 27MHz*16 = 432MHzAnother thing you could
try (at around line 1035) is to change value 0x40 into
0x00:SendRegister(16+0, 0x0C|3|0x00); // CLK0: 0x0C=PLLA local msynth;
3=8mA; 0x40=integer division;
bit7:6=0->power-upSendRegister(16+1, 0x0C|3|0x00); // CLK1: 0x0C=PLLA local
msynth; 3=8mA; 0x40=integer division;
bit7:6=0->power-upSendRegister(16+2, 0x2C|3|0x00); // CLK2: 0x2C=PLLB local
msynth; 3=8mA; 0x40=integer division; bit7:6=0->power-upWill you let me
know if it fixing your issue (or not) ?  *

I see a constant noisefloor of -112 dBm on the signal meter

With Firmware 1.02m, the noisefloor is -89 dBm and more "birdies" and plops
as I earlier described are back.
I have been playing around 80m and 40m with FW1.02i and was very surprised
by the performance, so for now I go back to 1.02i

threeme3, 2020-10-13:
Hi Cees,

Here is a new version, where the VOX control is now entirely working from the menu (3.1).
No need to use the long-right button press anymore :-).
When switched ON, you can freely tune around and control the rig and the s-meter keeps working (as long in RX).
The new version is here: https://github.com/threeme3/QCX-SSB/tree/feature-rx-improved

Regarding the pops, it should also be resolved now. The s-meter is now calibrated, but might still indicate more signal as before.

I am interested to know if you think R1.02m is better now than R1.02i.

73, Guido

PE1DDA, 2020-10-15:
Hello Guido,

Thnx for your kind reply
Just flashed your latest R1.02m from GitHub after building a more stable
flashing unit.
Vox (and) menu work like a charm, I like it better this way, also the menu
in general has improved because if active it can be clearly seen by
the *> *followed
by value to be changed.
S meter keeps functional while on Vox, love that too (for whatever
reason),  noise floor on this 1.02m and 50 Ohm termination = -125dBm
Plops are gone and birdies are less.
Is it my illusion or did you also change something in the AGC function?
I also noticed the PA Bias max setting (on a reset) by default is set to
128, good thing and much safer than the 220 it was once before especially
when experimenting with HW as I like to do
So, a big thank you!
I really like your work, that written and said, a question on behalf of
more users in the group using the rig for FT8

On all developed boards, the PTT and MIC are both on one 3,5 mm
jack-socket. Most people like me, will use a standard 3,5 mm stereo
PC/Audio cable from
the junkbox or Action to connect to their PC or Laptop.
Tip = mic
Ring = PTT
Ground

Doing this will, in most cases, activate the PTT switch if the cable is and
stays connected to the soundcard and the uSDX is switched on/off, due to a
low impedance on the PTT pin, thus causing a total reset
Is it possible to deactivate the PTT function in SW and make it an option
before flashing like something that can be opted in or out as you did with
the rotary encoder direction.

Love to read from you, all the best!

threeme3, 2020-10-15:
Hi Cees,

Thank you for the feedback. Good to hear that the VOX is working well.
You are correct that also the AGC function was fixed.

Regarding the factory reset when PTT is keyed on startup; this already has been changed.
Now, a factory reset can be induced by pressing the right-button on startup (not longer the case with PTT).
I have now made another change where keying the PTT on startup, will show a flashing T and prevents the transceiver to be in TX mode for safety reasons.

73, Guido

closed 2020-10-15 by threeme3