threeme3/usdx #28: Split out ino file into multi files

howard0su, 2020-10-09:
It is inconvenient to maintain such big file. Also this limit us to build the different firmware for the different board by default. I would suggest to do the following folder structure to organize the code instead:

src/*.h -> the main source code. We can split the code into multi .h files like cat.h, radio.h, menu.h, globals.h, lcd1602.h, si5351.h, led.h, debug.h
QCX-SSB/QCX-SSB.ino  -> the build folder to build QCX-SSB firmware. This ino contains the definition and a serial of includes to include the files in src/
uSDX/uSDX.ino -> the build folder to build uSDX firmware. 
uSDX_ABC/uSDX_ABC.ino -> another build for another board.

Thoughts?

howard0su, 2020-10-10:
lcd may need several files for different lcd option like 1602 and LED.

krzysztof9nowak, 2021-06-21:
Hello,
I would like to contribute to the project. I think we can use solutions already proven in a big and popular Arduino project - [Marlin 3D Printer Firmware](https://github.com/MarlinFirmware/Marlin). They are using PlatformIO, which allows for developing in a real IDE like VS Code, while still keeping the reverse compatibility with a simple Arduino IDE.

Regarding compiling for different boards, a well thought out system of header files, each providing a configuration for one type of board might be better than having multiple .ino entry points.

jokubasver, 2021-09-10:
> Hello,
> I would like to contribute to the project. I think we can use solutions already proven in a big and popular Arduino project - [Marlin 3D Printer [...]

PlatformIO is a great idea, I find it much more convenient and easier to use than Arduino IDE. 
And I see that you have it split out the project into multiple files: https://github.com/krzysztof9nowak/QCX-SSB/tree/feature-platformio

I haven't tried it, but I assume it works fine.

This could be a good starting point for Guido to implement this into his repo.

W3JDR, 2021-09-18:
I'd like to suggest that the functions related to the receiver DSP processing be seperated out. This way, someone who wants to build a variant with a different processor can do it. I think the TX stuff should be separated too, as I see the present approach as a laboratory curiosity that has limited applicability, even though it is quite clever. 
Joe