threeme3/usdx #32: Support WSPR

howard0su, 2020-10-10:
It now has some ROM space to add the support to transit WSPR beacons.

We can create a simple webpage to encode the data of the callsign, position, 5w for WSPR. This will creates a symbol with 162 uint8_t array. put it into the code and compile the firmware.

Introduce a new mode called WSPR to transmit. The precise timestamp can be feed by human by click the button to give a precise start time. Then the machine can measure the time.  Later on, we can add the support reading from GPS. 

WSPR mode can have the following parameters:
1. idle time -- how much slot to become idle.
2. bands -- transit on how many bands
3. bands random - random select bands or one-by-one order.

Thoughts? I need some help to guide me how to do the transition of WSPR symbols.

howard0su, 2020-10-10:
I find this project is interesting:
http://wsprnet.org/drupal/sites/wsprnet.org/files/si570wspr.pdf

```
>wsprcfg k1jt 1 52 2 Symbols for callsign= K1JT ;lon=1;lat=52;dbm=27 3,3,2,0,2,2,2,2,1,0,2,0,3,3,3,2,2,0,3,2,0,3,0,3,1,1,3,0,0,2,2,0,0,0,1,0,0,3,2,3, 2,2,2,0,0,0,3,0,1,3,2,0,3,1,2,1,0,0,0,3,1,2,1,2,2,2,0,1,3,2,1,2,3,0,1,0,1,0,0,3, 2,0,1,0,1,3,2,2,0,3,3,0,3,0,1,0,2,2,3,0,2,0,0,2,1,0,0,3,2,0,1,1,1,0,1,3,2,0,1,3, 2,1,2,2,2,1,1,3,2,2,2,0,0,1,2,3,2,0,3,1,0,0,0,0,2,2,0,3,3,0,1,0,3,3,2,2,2,1,3,0, 2,2
```

kholia, 2020-10-11:
> We can create a simple webpage to encode the data of the callsign, position, 5w for WSPR.

@howard0su Hey - I have created such a page now.

https://kholia.github.io/wspr_encoder.html

Thanks for writing so many nice patches for the QCX-SSB repository <3

howard0su, 2020-10-11:
Nice work. 

Why the compressed version has so many zero in the end?

and 4FSK only 0,1,2,3 so we can further compress 4 symbol into one byte. So we only need 41bytes.

kholia, 2020-10-11:
> Why the compressed version has so many zero in the end?

The compression loop was terminating prematurely - fixed now. Thanks!

Neat idea. I am implementing this 41-bytes compression scheme now.

kholia, 2020-10-11:
The current compression code is simple:

```js
var len = 162;
var compressed_output = new Uint8Array(len/2);

function compress(buffer) {
        var idx = 0;
        for (var i = 0; i < len; i = i + 2) {
                compressed_output[idx] = (buffer[i] << 4) + (buffer[i+1]);
                idx = idx + 1;
        }
}
```

kholia, 2020-10-11:
> I am implementing this 41-bytes compression scheme now.

https://kholia.github.io/wspr_encoder.html has been updated with this scheme now.

It now decompresses the compressed data for verification purposes. This should catch silly bugs.

howard0su, 2020-10-13:
Is there is a tail zero in the output of "More Compressed Output (162 4FSK symbols in 41 bytes):"

kholia, 2020-10-13:
The following original 162 bytes are zero-padded to length 164.

`3,3,0,2,0,2,0,0,1,2,0,2,3,3,1,0,2,2,1,2,0,1,0,1,3,3,3,0,0,2,2,2,0,2,1,0,0,3,0,3,2,2,0,0,0,0,3,0,3,1,2,0,1,3,0,3,0,2,0,1,1,2,3,2,2,0,0,3,1,0,1,2,3,0,3,2,1,2,0,3,2,0,1,0,1,3,0,0,0,1,1,0,1,2,3,2,0,2,1,0,2,0,0,0,1,0,0,3,0,2,1,3,1,2,3,3,0,0,1,1,2,3,0,0,2,3,3,3,2,2,2,2,2,3,2,3,0,2,1,3,0,2,2,0,0,2,2,3,1,0,3,2,1,3,2,2,2,3,3,2,0,0`

So we have 4 zeroes at the end to compress. This encodes to single compressed `0` at the end. During decompression, we expand this single compressed zero back to 4 zeroes and then we ignore the 2-bytes (zero pad) at the end.

The decompression results (should) match the original 162 bytes.

The compression and decompression code is straightforward.

```js
var actual_length = 162;
var len = actual_length + 2;  // 162 symbols + zero-pad at end
var more_compressed_output = new Uint8Array(len/4);
var decompressed_output = new Uint8Array(len);

function decompress(buffer) {
        var idx = 0;
        for (var i = 0; i < len/4; i = i + 1) {
                decompressed_output[idx] = (buffer[i] & 0xC0) >> 6;
                decompressed_output[idx+1] = (buffer[i] & 0x30) >> 4;
                decompressed_output[idx+2] = (buffer[i] & 0x0C) >> 2;
                decompressed_output[idx+3]= (buffer[i] & 0x03);
                idx = idx + 4;
        }
}

function compress_more(buffer) {
        var idx = 0;
        for (var i = 0; i < len; i = i + 4) {
                more_compressed_output[idx] = ((buffer[i] << 6) + (buffer[i+1] << 4) + (buffer[i+2] << 2) + buffer[i+3]);
                idx = idx + 1;
        }
}

var buf = Module._malloc(len);
aresult = Module.ccall('real_main', // name of C function
'number', // return type
['string','string','string','number'], // argument types
[callsign, grid, power, buf]); // arguments
console.log("Status (0 -> OK) is " + aresult);

var data = new Uint8Array(Module.HEAPU8.buffer, buf, len).slice(0, actual_length);
var output = data.join(",");
// console.log(output);
document.getElementById("output").value = output;
compress_more(data);
var output = more_compressed_output.join(",");
document.getElementById("optimized_output").value = output;
decompress(more_compressed_output);
var output = decompressed_output.slice(0, actual_length).join(",");
document.getElementById("decompressed_output").value = output;
```

maqifrnswa, 2021-03-16:
Also see https://github.com/etherkit/JTEncode for a WSPR implementation on an atmega328 arduino platform using an si5351. Should be able to drop in and use.

kholia, 2021-07-28:
https://github.com/kholia/uSDX_Beacon does WSPR and FT8 beacons now. However, it is an alternate dedicated beacon firmware created for HF PA testing, and development purposes.