threeme3/usdx #49: Switching off automatic decoder CW

ok1if, 2021-03-12:
In the latest version, the CW automatic decoder cannot be switched off. This is very annoying for us, what we can do CW better than that decoder. Please return the shutdown. Thank you. Milan OK1IF

threeme3, 2021-03-14:
Milan, 
Yes,! High-speed CW works better without the CW decoder.
The CW decoder should be visible in the menu again. Thanks for reporting.
Guido

closed 2021-03-14 by threeme3

ok1if, 2021-03-14:
HI Guido
 
With the latest issue, I cant to set in menu 8,3 my Ref freq.
 
Here is my setting:
 
// Configuration switches; remove/add a double-slash at line-start to enable/disable a feature; to save space disable e.g. CAT, DIAG, KEYER
#define DIAG            1   // Hardware diagnostics on startup (only disable when your rig is working)
#define KEYER           1   // CW keyer
//#define CAT             1   // CAT-interface
#define F_XTAL 25000165   // 25MHz SI5351 crystal  (enable for WB2CBA-uSDX or SI5351 break-out board)
//#define F_XTAL 27005000     // 27MHz SI5351 crystal
//#define F_XTAL 25004000   // 25MHz SI5351 crystal  (enable for WB2CBA-uSDX or SI5351 break-out board)
//#define F_XTAL 25000000   // 25MHz SI5351 crystal  (enable for 25MHz TCXO)
#define SWAP_ROTARY   1   // Swap rotary direction (enable for WB2CBA-uSDX)
//#define QCX           1   // Supports older (non-SDR) QCX HW modifications (QCX, QCX-SSB, QCX-DSP with I/Q alignment-feature)
//#define OLED          1   // OLED display, connect SDA (PD2), SCL (PD3)
//#define LPF_SWITCHING_DL2MAN_USDX_REV3         1   // Enable 8-band filter bank switching: latching relays wired to a TCA/PCA9555 GPIO extender on the PC4/PC5 I2C bus; relays are using IO0.0 as common (ground), IO1.0..7 used by the individual latches K0-7 switching respectively LPFs for 10m, 15m, 17m, 20m, 30m, 40m, 60m, 80m
//#define LPF_SWITCHING_DL2MAN_USDX_REV2       1   // Enable 5-band filter bank switching: latching relays wired to a TCA/PCA9555 GPIO extender on the PC4/PC5 I2C bus; relays are using IO0.1 as common (ground), IO0.3, IO0.5, IO0.7, IO1.1, IO1.3 used by the individual latches K1-5 switching respectively LPFs for 20m, 30m, 40m, 60m, 80m
//#define LPF_SWITCHING_DL2MAN_USDX_REV2_BETA  1   // Enable 5-band filter bank switching: latching relays wired to a PCA9539PW   GPIO extender on the PC4/PC5 I2C bus; relays are using IO0.1 as common (ground), IO0.3, IO0.5, IO0.7, IO1.1, IO1.3 used by the individual latches K1-5 switching respectively LPFs for 20m, 30m, 40m, 60m, 80m
//#define LPF_SWITCHING_DL2MAN_USDX_REV1       1   // Enable 3-band filter bank switching: latching relays wired to a PCA9536D    GPIO extender on the PC4/PC5 I2C bus; relays are using IO0 as common (ground), IO1-IO3 used by the individual latches K1-3 switching respectively LPFs for 20m, 40m, 80m
 
// Advanced configuration switches
//#define CAT_EXT       1   // Extended CAT support: remote button and screen control commands over CAT
//#define CAT_STREAMING 1   // Extended CAT support: audio streaming over CAT, once enabled and triggered with CAT cmd, 7812sps 8-bit unsigned audio is sent over UART. The ";" is omited in the data-stream, and only sent to indicate the beginning and end of a CAT cmd.
#define CW_DECODER      1   // CW decoder
#define TX_ENABLE       1   // Disable this for RX only (no transmit), e.g. to support uSDX for kids idea: https://groups.io/g/ucx/topic/81030243#6276
//#define TX_DELAY        1   // Enables a delay in the actual transmission to allow relay-switching to be completed before the power is applied
#define KEY_CLICK       1   // Reduce key clicks by envelope shaping
#define SEMI_QSK        1   // Just after keying the transmitter, keeps the RX muted for a short amount of time in the anticipation for continued keying
#define RIT_ENABLE      1   // Receive-In-Transit alternates the receiving frequency with an user-defined offset to compensate for any necessary tuning needed on receive
#define VOX_ENABLE      1   // Voice-On-Xmit which is switching the transceiver into transmit as soon audio is detected (above noise gate level)
//#define MOX_ENABLE    1   // Monitor-On-Xmit which is audio monitoring on speaker during transmit
//#define VSS_METER     1   // Supports Vss measurement (as s-meter option)
//#define SWR_METER     1   // Support SWR meter with bridge on A6/A7 (LQPF ATMEGA328P) by Alain, K1FM, see: https://groups.io/g/ucx/message/6262 and https://groups.io/g/ucx/message/6361
//#define CW_FREQS_QRP  1   // Defaults to CW QRP   frequencies when changing bands
//#define CW_FREQS_FISTS  1 // Defaults to CW FISTS frequencies when changing bands
//#define ONEBUTTON     1   // Use single (encoder) button to control full the rig; optionally use L/R buttons to completely replace rotory encoder function
//#define F_MCU_16MHZ   1   // 16MHz ATMEGA328P crystal  (enable for unmodified Arduino Uno/Nano boards with 16MHz crystal)
#define DEBUG         1   // for development purposes only (adds debugging features such as CPU, sample-rate measurement, additional parameters)
//#define TESTBENCH     1   // Tests RX chain by injection of sine wave, measurements results are sent over serial
 
73
Milan

> Od: "threeme3" ***@***.***>
> Komu: "threeme3/QCX-SSB" ***@***.***>
> Datum: 14.03.2021 11:00
 
Milan,
 Yes,! High-speed CW works better without the CW decoder.
 The CW decoder should be visible in the menu again. Thanks for reporting.
 Guido

threeme3, 2021-03-14:
OK, this is also fixed now.
Thanks again.
I suggest to disable DEBUG, as it makes things faster.
Guido

ok1if, 2021-03-14:
Hi Guido. I'm sorry, but even in the latest version I can't set REF FREQ 8.3 to the value I need. To the value 25000165. After uploading with #define F_XTAL 25000165, the value 14000000 will appear in menu 8.3. This cannot be entered manually, because it is not possible to change the step to higher orders.
Milan

> Od: "threeme3" ***@***.***>
> Komu: "threeme3/QCX-SSB" ***@***.***>
> Datum: 14.03.2021 15:29
 
OK, this is also fixed now.
 Thanks again.
 I suggest to disable DEBUG, as it makes things faster.
 Guido

threeme3, 2021-03-14:
Milan,
Can you try to reset the settings by keeping the rotary button pressed when the rig is starting up?

ok1if, 2021-03-14:
Guido, I'm not a programmer. I program the circuit by using an Arduino UNO into which I load the bootloader and then normally load the program using the Arduino UNO board. I tried to load the bootloader into the circuit again and reload the circuit and now everything is OK. I'm sorry for my mistake. Procedure with reset setting is also full OK.
Thank you
Milan

> Od: "threeme3" ***@***.***>
> Komu: "threeme3/QCX-SSB" ***@***.***>
> Datum: 14.03.2021 20:12
 
Milan,
 Can you try to reset the settings by keeping the rotary button pressed when the rig is starting up?

ok1if, 2021-03-14:
After reset all OK.
 
Otherwise Guido. I am a passionate CW operator and I like to broadcast from nature using QRP. I'm quite lacking that uSDX doesn't have the ability to upload at least one CW message. For example, a CQ. It would be enough to store 60 letters in memory. After playing, it would be excellent if the option to send the message again after a set time (eg 1 to 60 sec). Touching the paddle would then be able to stop this loop (giving CQ). Is there still room for this?
Starting sending message for example using contact encoder a together Dit on paddle. Some programs using two memories using dit and dot.
 
Thank you
Milan OK1IF

> Od: "threeme3" ***@***.***>
> Komu: "threeme3/QCX-SSB" ***@***.***>
> Datum: 14.03.2021 20:12
 
Milan,
 Can you try to reset the settings by keeping the rotary button pressed when the rig is starting up?

threeme3, 2021-03-15:
Good it is fixed.

In the latest code, I have added an experimental CW send message. Before uploading change the text at line 29 to your needs:
#define CW_MESSAGE_TEXT "CQ CQ DE UR0CAL UR0CAL K"

Currently only a single message can be sent once, when double click left button.

I will think about adding a feature to edit message in menu and to repeat the message as you described.

73, Guido

ok1if, 2021-03-15:
Hi Guido. Thank you for the latest version with CW memory. It will probably send a message, but there is no CW speed rate setting in the menu. Furthermore, manual paddle keying does not work.
73
Milan

> Od: "threeme3" ***@***.***>
> Komu: "threeme3/QCX-SSB" ***@***.***>
> Datum: 15.03.2021 13:12
 
Good it is fixed.
In the latest code, I have added an experimental CW send message. Before uploading change the text at line 29 to your needs:
 #define CW_MESSAGE_TEXT "CQ CQ DE UR0CAL UR0CAL K"
Currently only a single message can be sent once, when double click left button.
I will think about adding a feature to edit message in menu and to repeat the message as you described.
73, Guido

threeme3, 2021-03-15:
Milan,
Compiling with KEYER enabled should give you the CW speed rate and give you manual padding.
It is currently only possible to paddle once the CW message is completed, there is currently no way to stop it.

ok1if, 2021-03-15:
Guido
I have:
 
#define KEYER           1   // CW keyer
 
But sorry not answer for paddle.
 
Milan

> Od: "threeme3" ***@***.***>
> Komu: "threeme3/QCX-SSB" ***@***.***>
> Datum: 15.03.2021 17:31
 
Milan,
 Compiling with KEYER enabled should give you the CW speed rate and give you manual padding.
 It is currently only possible to paddle once the CW message is completed, there is currently no way to stop it.

threeme3, 2021-03-15:
Milan,
Not sure what is going on.
With KEYER enabled you should see menu option 2.6 Keyer Mode and be able to set it to Iambic A for paddle, and see also 2.5 Keyer Speed.
I will finish some parts and let you know when it is ready.

ok1if, 2021-03-15:
I'm sorry Guido. I probably messed up something while copying. Now I've compiled everything again and everything is fine. I just had to do a reset setting, due to the wrong frequency value. If the automatic decoder is switched on, then the display shows the letters of what is being manually keyed at that moment. Maybe this string could be stored in memory. This would eliminate the need to edit CW memory. If necessary, the text would be reloaded. This is how the keying circuit from N0XAS solves it. In short, in the menu item "load CW memory". N0XAS ends the recording with a special command
-...-. (pseudo BN). This is also a command to start a retry. He uses another variable "B" for the repetition time, which is in the range 1-60 sec). I think for a CW operator he could easily give for example -...-. (pseudo BN) followed by a number (eg 1-60), which would be the waiting time for repetition. If I want to end this, I touch the keys on the DIT and I jump out of the loop and I can put it manually over the paddle.
Once again, I apologize for the confusion I sometimes create.
Milan

> Od: "threeme3" ***@***.***>
> Komu: "threeme3/QCX-SSB" ***@***.***>
> Datum: 15.03.2021 18:17
 
Milan,
 Not sure what is going on.
 With KEYER enabled you should see menu option 2.6 Keyer Mode and be able to set it to Iambic A for paddle, and see also 2.5 Keyer Speed.
 I will finish some parts and let you know when it is ready.

threeme3, 2021-03-16:
Milan,
There is version R1.02p that where you can enable CW_MESSAGE on line 28.
Menu item 2.9 and 2.10 sets a message and controls the interval (0 = no repetition).
TX stops as soon a button or key is pressed. CQ message can be restarted by double left-click.
73, Guido

ok1if, 2021-03-16:
Super Guido I tryed it, but sorry in MENU is not setting for paddle JAMBIC A JAMBIC B and STRAIGHT KEY. It is unpossible to use paddle.
Milan

> Od: "threeme3" ***@***.***>
> Komu: "threeme3/QCX-SSB" ***@***.***>
> Datum: 16.03.2021 17:37
 
Milan,
 There is version R1.02p that where you can enable CW_MESSAGE on line 28.
 Menu item 2.9 and 2.10 sets a message and controls the interval (0 = no repetition).
 TX stops as soon a button or key is pressed. CQ message can be restarted by double left-click.
 73, Guido

ok1if, 2021-03-16:
Sorry again Guido.
All is OK and full function. Again my mistake. I dont know why, but I must every time to make reset setting. Congratulation. Super Job
Milan

> Od: "threeme3" ***@***.***>
> Komu: "threeme3/QCX-SSB" ***@***.***>
> Datum: 16.03.2021 17:37
 
Milan,
 There is version R1.02p that where you can enable CW_MESSAGE on line 28.
 Menu item 2.9 and 2.10 sets a message and controls the interval (0 = no repetition).
 TX stops as soon a button or key is pressed. CQ message can be restarted by double left-click.
 73, Guido

ok1if, 2021-03-17:
Hi Guido.
First of all, I must state that the request for CW memory was successfully completed. As a telegraph operator, I prefer to record text using a paddle (it's much faster), on the other hand, I realize that for a beginner, the system you use is better. This system is also advantageous for lighthouse text. However, I have a question whether the maximum length of the text is somehow programmatically limited. Sometimes, when wandering in the woods, I use beacon broadcasts to find out what I'm doing. So I think that the maximum length of the beacon text up to 256 letters is quite sufficient. I also wanted to ask about the recommended settings for SSB broadcasting. I use TX Drive 4 myself, PA BIAS min 5, PA BIAS max 128. Nevertheless, the comment of the counter stations is that it growls. Unfortunately, nothing can be verified on the slave receiver, because it always seems "very bad". The lower the level of the received signal, the better. However, that is not my main problem. I understand the limitations of Dynamic Range. I am rather bothered by the insufficient suppression of the second sideband during reception. This is only max 25dB for my 4pcs device, even after careful balancing of the IQ Phase. If I / Q amplitude could be balanced, then I think it would be easy to achieve values ​​around 35dB (and more), which is guaranteed to be sufficient for practical operation (my worst home made TCVR with simple crossbar filters reach this value and I quite satisfied). Another thing is the use of split operation. Although I have the option to set VFO A and VFO B, the SPLIT item is missing, where I listen to VFO A and broadcast on VFO B. In this mode, there is no need to think about RIT and I would easily ban it. Of course, the more expensive TCVR can continue to work with these two VFOs. However, I understand that it unnecessarily complicates programming. So in the first phase it would be enough to just select the item SPLIt in the menu (I'm broadcasting on B and listening on A). A skilled operator would then find his way to do with it. It's actually similar to working with RITE. To admit, I never believed that SDX would be usable. Now AGC also works (fast AGC on CW is an excellent thing). DSP filters also work. Maybe I don't quite understand what NR is for CW, when it rather has a negative effect. Sometimes I also suffer from whistling at the reception (as if something wanted to vibrate). But if one realizes that everything is happening on an 8-bit processor with 10-bit A / D conversion, then it is actually a miracle. This radio on the WB2CBA board (I use version 1.01) can be assembled in 4.5 hours and it is actually the cheapest and highest quality QRP TCVR I know. Congrats again.
Milan
OK1IF

> Od: "threeme3" ***@***.***>
> Komu: "threeme3/QCX-SSB" ***@***.***>
> Datum: 16.03.2021 17:37
 
Milan,
 There is version R1.02p that where you can enable CW_MESSAGE on line 28.
 Menu item 2.9 and 2.10 sets a message and controls the interval (0 = no repetition).
 TX stops as soon a button or key is pressed. CQ message can be restarted by double left-click.
 73, Guido