threeme3/usdx #58: CW message auto tx broken in 1.2w

mautas, 2021-11-14:
I'm trying out the CW auto tx function in 1.2w. It doesn't work, just shows a 0 in the display and exits. I don't know if this is not yet implemented in this version (a note in the definition header would have saved me and other people some time) or if it was working but it was seriously broken in this version.

threeme3, 2021-12-01:
Yes it should work. Please try to do a factory reset by holding the encoder button while powering it up, and try again.

mautas, 2021-12-01:
It doesn't, I checked out the code and all it does when it enters the CW
message Tx loop is print a 0 in the screen and then exits the loop. I tried
to understand the code (ie. why does it leave the loop so fast) but its
kind of hard to understand how it works.
ᐧ

threeme3, 2021-12-01:
Ok, the CW message will stop as soon a button or a dit/dah I sent. Maybe there is RF feedback on one of these inputs, interrupting the CW Message process.