threeme3/usdx #60: S- meter zero/calibrating

kefeur, 2021-12-08:
The S meter shows a value of 4 when the input is closed. I would like to ask how to set it to 0 or calibrate it.
Is it possible at all?
Thank you!
Ferenc.

kefeur, 2021-12-10:
-

threeme3, 2021-12-12:
Hi Ferenc,
Normally there is no need to calibrate the s-meter. You are saying that you closed the input, what do you mean by that? You also did not tell me what usdx design you are using. Shortening the adc inputs will not work as they needs to be biased with 1.65v. Did you put a 50ohm resistor on the antenna input? In such case the s-meter should fall to the minimum noise level, which is should be something  around -109dbm up lo -124dbm depending on the op amps used, and how well the design is. If you see deviating values, it could be that there is some hardware issue, check if the ADC bias is ok and if both opamp stages are operational. 
Good luck.

kefeur, 2021-12-13:
Thanks for the reply!
Board version "TRX V1.02" , Firmware 1.02w

The voltage at the ADC inputs is good.
Antenna input closed at 50ohm.
If I remove the opamp ot it will still display -95 dbm.
I also see that the L and R buttons don't work well either.
I also replaced atmega328 by reprogramming the same error ...
Fuse bits are also set well.

kefeur, 2021-12-14:
I tested it with an arduino pro mini, the pushbuttons work very well and the display shows -107pcs.
This is probably not the luckiest type of trx motherboard.
My appreciation for the project is very imaginative and well executed!
Congratulations to.

threeme3, 2021-12-14:
Yes, you checked the right things on the TRX V1.02 board. Maybe there is an issue with the power supply, is it rectified properly? and if so, check there electrolytic capacitors if they are working well. With -95dBm on the s-meter even when opamp is not there, suggests that there is either something wrong with cleanness if the voltage supply. If you use an OLED display I suggest to apply the WB2CBA oled mod to make it less noisy. Good luck in finding the cause and the rest of the build.
73, Guido

kefeur, 2021-12-14:
Thank you for the encouragement.
Keep up the good work...
73 'Ferenc

closed 2021-12-14 by kefeur