threeme3/usdx #63: Internal Pull-Up issue with minicore

armSeb, 2022-01-19:
Hi! 

First,  thank you for this project, it's really cool!

I am playing with your code and I have an issue with the INPUT_PULLUP.

When I use MiniCore (version 2.1.3 in the arduino IDE 1.8.16), the PD7 and PD7 lines are not pulled up. This causes the rotary encoder not working. If I add two 10k resistors the rotary encoder works well. 
If I use the regular arduino Uno board (in the IDE), the pull-up function works fine. But I encounter another issue : the pull-up is applied on all of the input, especially on the PC3 input. I get about 1,2V when no button is pressed. This causes an issue with the voltages thresholds.

Did you already have this behavior ?

threeme3, 2022-01-19:
Hi Sebastien,

Thanks for reporting this issue. I have never seen such behavior before, and have no idea what this could be caused by.

First question I have is, are all VCC pins connected to 5V (for a tqfp package there are two VCC inputs). Second question I have, did you try to built with Arduino Uno board settings and are the issues the same? Third question, could it be a clone of atmega328p chip which does not implement the pull-up well, or could it be damaged?

armSeb, 2022-01-19:
Hi,

1) Yes I checked, and all VCC and GND pins are connected. 

2) My second try is with the Arduino Uno settings, and the issue is reversed : I have the pull-up on the encoder pins and even on the pin with no configured pull-up. 

3)  This is a possibility, It's a crappy chinese device. I am modifying it with better quality components (especially the toroidals inductors). It worked fine before I flashed the firmware (but with problematic AM and FM rx). This issue remains with the latest software. I can receive AM in the USB or LSB mode but not in AM mode when I am centered exactly on the frequency. I checked the I/Q signals with an oscilloscope and they looks fine (same amplitude, and 90 degrees between them). Maybe its a normal behavior as this device is designed for SSB.

threeme3, 2022-01-20:
Hi Sebastien,
It could also be an issue with minicore, not properly mapping I/O registers. Did you try with naive Arduino Uno compilation?

armSeb, 2022-01-20:
The sketch works well with the native Arduino Uno compilation.

I just tried an empty sketch with only : 

`pinMode(7, INPUT);
  pinMode(6, INPUT_PULLUP);`

And it works well with both the native Arduino and  the minicore bootloader. Maybe there is a conflict between minicore and something in this sketch. I'll keep you updated if I found anything about it. Thank you :)